bluez/bluer #2: Can't compile libbluetooth

bobgates, 2021-07-06:
Lots of errors like:

```pub const RFCOMMGETDEVINFO: u64 = request_code_read!('R', 211, std::mem::size_of::<c_int>());
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `u64`, found `u32` 
```

when building on Raspberry Pi running:
Linux raspberrypi 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux

The errors occur when trying to build libbluetooth. I guess that's why there is a an existing issue to remove the dependency?

surban, 2021-07-06:
Looks like a 32-bit issue. I've been successfully compiling libbluetooth on RPi but on 64 bit.

For now you can disable the l2cap feature if you don't need it. But we plan to remove this dependency anyway so this will get fixed eventually.

bobgates, 2021-07-06:
Makes sense - stuff being 32 bit by default on a 32-bit OS.
 
Thanks for the tip, I removed l2cap and got a lot further. I then hit the issue with depending on the Nordic bluetooth numbers database. I copied those files into my tree, and it's building now.

closed 2021-07-06 by bobgates