bluez/bluer #5: Discovery example = "Invalid arguments in method call"

bobgates, 2021-07-07:
5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux
Using bluez 5.50. 

I'm trying to run the discover_devices example. I get this error:
```
Error: Error { kind: InvalidArguments, message: "Invalid arguments in method call" }
```

The code fails on this line in adapter.rs, in the function discovery_session, around about line 140 (I've been messing with code, line number is vague):
```
                    self.call_method("SetDiscoveryFilter", (filter.to_dict(),)).await?;
```

The filter it is called with contains:
```
DiscoveryFilter { uuids: {}, rssi: None, pathloss: None, transport: Auto, duplicate_data: false, discoverable: false, pattern: None }
```

surban, 2021-07-07:
You can set the log level to `trace` to see D-Bus messages.

Could you try again with bluez 5.59 or master? I've never tested older versions.

bobgates, 2021-07-07:
How do I set bluez 5.59? apt tells me I'm at most recent version?

surban, 2021-07-07:
http://www.bluez.org/release-of-bluez-5-59/

Either update your distribution or build and install from source.

bobgates, 2021-07-07:
Okay, that worked: discover_devices is now running

closed 2021-07-07 by bobgates

fujexo, 2022-09-30:
Got the example working on Raspbian Bullseye with `bluez 5.55-3.1`. Thanks for the hint about `bluez` update!
```
root@terrariumpi:~# lsb_release -a 
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
root@terrariumpi:~# dpkg -l | grep bluez
ii  bluez                          5.55-3.1                         armhf        Bluetooth tools and daemons
ii  bluez-firmware                 1.2-4+rpt8                       all          Firmware for Bluetooth devices
```

This didn't work on Raspbian Buster with `bluez 5.50-1.2~deb10u2+rpt1`
```
root@terrariumpi:~# RUST_LOG=trace /home/pi/bletest_bluer                                                                                                                                                                                   
[2022-09-30T02:34:45Z TRACE bluer::session] Connected to D-Bus with unique name :1.124629                                                                                                                                                   
[2022-09-30T02:34:45Z TRACE mio::poll] registering event source with poller: token=Token(1), interests=READABLE                                                                                                                             
[2022-09-30T02:34:45Z TRACE bluer::session] Starting event loop for :1.124629                                                                                                                                                               
Discovering devices using Bluetooth adapater hci0                                                                                                                                                                                           
                                                                                                                                                                                                                                            
[2022-09-30T02:34:45Z TRACE bluer::adapter] /org/bluez/hci0: org.bluez.Adapter1.Powered := true                                                                                                                                             
[2022-09-30T02:34:45Z TRACE bluer::session] Starting new single session for /org/bluez/hci0                                                                                                                                                 
[2022-09-30T02:34:45Z TRACE bluer::adapter] /org/bluez/hci0: org.bluez.Adapter1.SetDiscoveryFilter ({"Discoverable": Variant(false), "UUIDs": Variant([]), "Transport": Variant("auto"), "DuplicateData": Variant(false)},)                 
[2022-09-30T02:34:45Z TRACE bluer::adapter] /org/bluez/hci0: org.bluez.Adapter1.SetDiscoveryFilter (...) -> Err(D-Bus error: Invalid arguments in method call (org.bluez.Error.InvalidArguments))                                           
[2022-09-30T02:34:45Z TRACE bluer] DBus error org.bluez.Error.InvalidArguments: Invalid arguments in method call                                                                                                                            
[2022-09-30T02:34:45Z TRACE mio::poll] deregistering event source from poller                                                                                                                                                               
[2022-09-30T02:34:45Z TRACE mio::poll] deregistering event source from poller                                                                                                                                                               
Error: Error { kind: InvalidArguments, message: "Invalid arguments in method call" }                                                                                                                                                        
```