bluez/bluer #6: Can't build code depending on bluer = "0.10.2"

laszlourszuly, 2021-07-16:
Gentlemen!

I'm seeing _"no method named `try_read_buf` found"_ issues when building my code, having included a dependency to bluer 0.10.2 (please find my config and build output with exact error message below).

I'm not trying to manually build the bluer library, but rather consume it from crates.io (i guess).

I do have a bit aged version of Bluez on my system (5.53 I believe), but I'm not convinced this is the real issue at this stage as I'm experiencing build issues rather than runtime issues.

_Development environment:_
* Ubuntu 20.04
* bluez 5.53-0ubuntu3.3
* libdbus-1-dev 1.12.16-2ubuntu2.1
* cargo 1.53.0 (4369396ce 2021-04-27)
* rustc 1.53.0 (53cb7b09b 2021-06-17)

_Cargo.toml:_
```
[package]
name = "hello"
version = "0.0.1-snapshot"
edition = "2018"

[dependencies]
bluer = "0.10.2"
```

_The critical part of `cargo build` output:_
```
   Compiling bluer v0.10.2
error[E0599]: no method named `try_read_buf` found for struct `tokio::net::UnixStream` in the current scope
   --> /home/monster/.cargo/registry/src/github.com-1ecc6299db9ec823/bluer-0.10.2/src/gatt/mod.rs:134:29
    |
134 |         let n = self.stream.try_read_buf(&mut buf)?;
    |                             ^^^^^^^^^^^^ help: there is an associated function with a similar name: `try_read`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0599`.
error: could not compile `bluer`
```

laszlourszuly, 2021-07-16:
I think I would qualify as a seasoned developer (15+ years in the business) but I am new to Rust, so I am humble to the possibility that the issue is a misconfiguration on my end.

Basically I'm wondering if this is a known issue and if you have any pointers on how i could get around it?

surban, 2021-07-18:
In fact we were missing a dependency crate feature and due to how Cargo works it does not show up as an error in CI.

It should be fixed in version 0.10.3.

laszlourszuly, 2021-07-19:
Thank you for your quick actions! I can confirm it working with version 0.10.3. Closing this issue.

closed 2021-07-19 by laszlourszuly