bluez/bluer #15: failed build

ChrisFetterly, 2021-12-27:
Hi, sorry am quite new to rust. I keep getting build errors and am unsure how to resolve.

```
(base) chrisfetterly@computer % cargo build
    Updating crates.io index
  Downloaded proc-macro2 v1.0.36
  Downloaded 1 crate (41.4 KB) in 0.55s
   Compiling proc-macro2 v1.0.36
   Compiling libc v0.2.112
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.84
   Compiling cfg-if v1.0.0
   Compiling memchr v2.4.1
   Compiling log v0.4.14
   Compiling futures-core v0.3.19
   Compiling autocfg v1.0.1
   Compiling slab v0.4.5
   Compiling pin-project-lite v0.2.7
   Compiling version_check v0.9.3
   Compiling futures-channel v0.3.19
   Compiling futures-task v0.3.19
   Compiling futures-sink v0.3.19
   Compiling futures-util v0.3.19
   Compiling pin-utils v0.1.0
   Compiling futures-io v0.3.19
   Compiling serde_derive v1.0.132
   Compiling pkg-config v0.3.24
   Compiling serde v1.0.132
   Compiling cfg-if v0.1.10
   Compiling unicode-segmentation v1.8.0
   Compiling bytes v1.1.0
   Compiling once_cell v1.9.0
   Compiling serde_json v1.0.73
   Compiling lazy_static v1.4.0
   Compiling signal-hook v0.3.13
   Compiling itoa v1.0.1
   Compiling bitflags v1.3.2
   Compiling parking_lot_core v0.8.5
   Compiling ryu v1.0.9
   Compiling scopeguard v1.1.0
   Compiling smallvec v1.7.0
   Compiling bytes v0.5.6
   Compiling pin-project-lite v0.1.12
   Compiling async-trait v0.1.52
   Compiling hashbrown v0.11.2
   Compiling byteorder v1.4.3
   Compiling ppv-lite86 v0.2.15
   Compiling termcolor v1.1.2
   Compiling regex-syntax v0.6.25
   Compiling unicode-width v0.1.9
   Compiling humantime v2.1.0
   Compiling hex v0.4.3
   Compiling strsim v0.10.0
   Compiling pretty-hex v0.2.1
   Compiling instant v0.1.12
   Compiling lock_api v0.4.5
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling unicase v2.6.0
   Compiling memoffset v0.6.5
   Compiling indexmap v1.7.0
   Compiling num-traits v0.2.14
   Compiling textwrap v0.14.2
   Compiling heck v0.3.3
   Compiling libdbus-sys v0.2.2
   Compiling aho-corasick v0.7.18
   Compiling os_str_bytes v4.2.0
   Compiling quote v1.0.10
   Compiling regex v1.5.4
   Compiling signal-hook-registry v1.4.0
   Compiling num_cpus v1.13.1
   Compiling mio v0.7.14
   Compiling getrandom v0.2.3
   Compiling iovec v0.1.4
   Compiling net2 v0.2.37
   Compiling atty v0.2.14
   Compiling nix v0.23.1
   Compiling bytes v0.4.12
   Compiling uuid v0.8.2
   Compiling rand_core v0.6.3
   Compiling parking_lot v0.11.2
   Compiling mio v0.6.23
   Compiling rand_chacha v0.3.1
   Compiling signal-hook-mio v0.2.1
   Compiling rand v0.8.4
   Compiling crossterm v0.22.1
   Compiling mio-uds v0.6.8
   Compiling env_logger v0.9.0
   Compiling tokio v0.2.25
   Compiling synstructure v0.12.6
   Compiling futures-macro v0.3.19
   Compiling tokio-macros v1.7.0
   Compiling pin-project-internal v1.0.9
   Compiling custom_debug_derive v0.5.0
   Compiling strum_macros v0.22.0
   Compiling displaydoc v0.2.3
   Compiling num-derive v0.3.3
   Compiling clap_derive v3.0.0-beta.5
   Compiling tokio v1.15.0
   Compiling custom_debug v0.5.0
   Compiling pin-project v1.0.9
   Compiling strum v0.22.0
   Compiling clap v3.0.0-beta.5
   Compiling dbus v0.9.5
   Compiling futures-executor v0.3.19
   Compiling futures v0.3.19
   Compiling tab-pty-process v0.2.0
   Compiling tokio-stream v0.1.8
   Compiling tokio-compat-02 v0.2.0
   Compiling dbus-tokio v0.7.5
   Compiling dbus-crossroads v0.5.0
   Compiling bluer v0.13.1 (/Users/chrisfetterly/Projects/RustProjects/bluer/bluer)
error[E0432]: unresolved imports `libc::SOCK_CLOEXEC`, `libc::SOCK_NONBLOCK`
 --> bluer/src/sock.rs:3:49
  |
3 | use libc::{c_int, c_ulong, sockaddr, socklen_t, SOCK_CLOEXEC, SOCK_NONBLOCK};
  |                                                 ^^^^^^^^^^^^  ^^^^^^^^^^^^^ no `SOCK_NONBLOCK` in the root
  |                                                 |
  |                                                 no `SOCK_CLOEXEC` in the root
  |
help: a similar name exists in the module
  |
3 | use libc::{c_int, c_ulong, sockaddr, socklen_t, O_CLOEXEC, SOCK_NONBLOCK};
  |                                                 ~~~~~~~~~
help: a similar name exists in the module
  |
3 | use libc::{c_int, c_ulong, sockaddr, socklen_t, SOCK_CLOEXEC, O_NONBLOCK};
  |                                                               ~~~~~~~~~~

error[E0432]: unresolved imports `libc::SOCK_CLOEXEC`, `libc::SOCK_NONBLOCK`
 --> bluer/src/gatt/mod.rs:5:22
  |
5 | use libc::{AF_LOCAL, SOCK_CLOEXEC, SOCK_NONBLOCK, SOCK_SEQPACKET};
  |                      ^^^^^^^^^^^^  ^^^^^^^^^^^^^ no `SOCK_NONBLOCK` in the root
  |                      |
  |                      no `SOCK_CLOEXEC` in the root
  |
help: a similar name exists in the module
  |
5 | use libc::{AF_LOCAL, O_CLOEXEC, SOCK_NONBLOCK, SOCK_SEQPACKET};
  |                      ~~~~~~~~~
help: a similar name exists in the module
  |
5 | use libc::{AF_LOCAL, SOCK_CLOEXEC, O_NONBLOCK, SOCK_SEQPACKET};
  |                                    ~~~~~~~~~~

error[E0432]: unresolved imports `libc::AF_BLUETOOTH`, `libc::SOL_BLUETOOTH`, `libc::TIOCINQ`
  --> bluer/src/l2cap.rs:21:5
   |
21 |     AF_BLUETOOTH, EAGAIN, EINPROGRESS, MSG_PEEK, SHUT_RD, SHUT_RDWR, SHUT_WR, SOCK_DGRAM, SOCK_SEQPACKET,
   |     ^^^^^^^^^^^^ no `AF_BLUETOOTH` in the root
22 |     SOCK_STREAM, SOL_BLUETOOTH, SOL_SOCKET, SO_ERROR, SO_RCVBUF, TIOCINQ, TIOCOUTQ,
   |                  ^^^^^^^^^^^^^ no `SOL_BLUETOOTH` in the root    ^^^^^^^ no `TIOCINQ` in the root

error[E0432]: unresolved imports `libc::AF_BLUETOOTH`, `libc::SOL_BLUETOOTH`, `libc::TIOCINQ`
  --> bluer/src/rfcomm/mod.rs:16:12
   |
16 |     c_int, AF_BLUETOOTH, EAGAIN, EINPROGRESS, MSG_PEEK, SHUT_RD, SHUT_RDWR, SHUT_WR, SOCK_RAW, SOCK_STREAM,
   |            ^^^^^^^^^^^^ no `AF_BLUETOOTH` in the root
17 |     SOL_BLUETOOTH, SOL_SOCKET, SO_ERROR, SO_RCVBUF, TIOCINQ, TIOCOUTQ,
   |     ^^^^^^^^^^^^^ no `SOL_BLUETOOTH` in the root    ^^^^^^^ no `TIOCINQ` in the root

error[E0425]: cannot find function `accept4` in crate `libc`
   --> bluer/src/sock.rs:147:15
    |
147 |         libc::accept4(socket.as_raw_fd(), saddr.as_mut_ptr() as *mut _, &mut length, SOCK_CLOEXEC | SOCK_NONBLOCK)
    |               ^^^^^^^ help: a function with a similar name exists: `accept`
    |
   ::: /Users/chrisfetterly/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.112/src/unix/mod.rs:610:5
    |
610 |     pub fn accept(socket: ::c_int, address: *mut sockaddr, address_len: *mut socklen_t) -> ::c_int;
    |     ----------------------------------------------------------------------------------------------- similarly named function `accept` defined here

Some errors have detailed explanations: E0425, E0432.
For more information about an error, try `rustc --explain E0425`.
error: could not compile `bluer` due to 5 previous errors
(base) chrisfetterly@computer bluer %```

surban, 2021-12-31:
Are you sure that you are building on Linux?

closed 2022-01-17 by surban

ouvanous, 2022-01-23:
Hi @surban, 
I am in the same situation. I try to use bluer on a Mac M1 and wonder if it is possible? 
I get the same error as above. 
Have a nice day 
Samuel

surban, 2022-01-24:
BlueZ is a Linux API. It cannot work on Mac OS.

ouvanous, 2022-01-24:
Ah ok! Many thanks