bluez/bluer #18: Resuming server notification sessions

mxk, 2022-01-11:
I'm implementing battery and HID peripheral services, which require caching of CCCD state across client connections. As far as I can tell, the current behavior for both `StartNotify` and `AcquireNotify` is that the notification session persists across client re-connections, but as soon as either the rust process or bluetoothd itself are restarted, there is no way to resume notifications. These services specify that the client is not required to re-enable notifications on each subsequent connection after bonding, so the client assumes that notifications will just work from there on.

I thought this would be controlled by the `Cache` option in `main.conf`, but that doesn't seem to be the case, at least not across bluetoothd restarts.

Is this a fundamental issue with the bluez API or is there some functionality that would enable this use case that isn't exposed in bluer?

surban, 2022-01-11:
BlueR exposes the [full BlueZ API with respect to GATT services](https://github.com/RadiusNetworks/bluez/blob/master/doc/gatt-api.txt). Thus this seems to be a limitation of the current BlueZ API and extensions to `bluetoothd` would be necessary to achieve the behavior you want.

mxk, 2022-01-11:
Understood, thanks for the quick response!

closed 2022-01-11 by mxk