bluez/bluer #22: ConnectDevice results in `org.freedesktop.DBus.Error.UnknownMethod`

ryankurte, 2022-01-27:
hi there,

attempting to use `Adapter::connect_device` results in:
```
Error { kind: Internal(DBus("org.freedesktop.DBus.Error.UnknownMethod")), message: "Method \"ConnectDevice\" with signature \"a{ss}\" on interface \"org.bluez.Adapter1\" doesn't exist\n" }
```

i have enabled experimental features in `bluetoothd`, and `d-feet` shows this method _does_ exist:

[image]

it _appears_ that `bluez` is expecting an `a{sv}` signature (a string:variant dictionary) while `bluer` is calling with `a{ss}` (a string:string dictionary). i had a go at patching this, but it also seems that `bluer` won't compile when included via a path ¯\\_(ツ)_/¯

```
error: failed to run custom build command for `bluer v0.13.2 (/home/ryan/projects/forks/bluer/bluer)`

Caused by:
  process didn't exit successfully: `/home/ryan/projects/spo2/target/debug/build/bluer-55d669f39f7d50ee/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=service_class_uuids.json
  cargo:rerun-if-changed=bluetooth-numbers-database/v1/service_uuids.json

  --- stderr
  thread 'main' panicked at 'services: Os { code: 2, kind: NotFound, message: "No such file or directory" }', /home/ryan/projects/forks/bluer/bluer/build.rs:227:6
```

relates to: https://github.com/deviceplug/btleplug/issues/244

surban, 2022-01-27:
Have you cloned Bluer with submodules?

surban, 2022-01-29:
Could you check again with the current master branch?

closed 2022-01-29 by surban

ryankurte, 2022-01-30:
> Have you cloned Bluer with submodules?

that's what i was missing, thanks!

> Could you check again with the current master branch?

the dbus call works nicely, though it turns out this method fails if the bluer already knows about the device so it actually doesn't seem to help with my device issue...

surban, 2022-01-31:
Calling `Adapter::remove_device` might help with such issues.

surban, 2022-02-02:
Have you tried the settings from the README for bluetoothd?

Also most problems are caused by old versions of BlueZ, not BlueR. So updating that might help.

ryankurte, 2022-02-03:
> Have you tried the settings from the README for bluetoothd?

only `--experimental`, i could try getting some more debug out though.

> Also most problems are caused by old versions of BlueZ, not BlueR. So updating that might help.

cheers, will have a look! though it strikes me as strange that `gattool` can talk to it (see linked btleplug issue) :-/

surban, 2022-02-03:
`gatttool` does not use the `bluetoothd`, so this indeed seems to be an issue with the daemon.

ryankurte, 2022-02-03:
> gatttool does not use the bluetoothd, so this indeed seems to be an issue with the daemon.

aha, that's super helpful thanks! i'll have to see if i can get the debian testing release running on my bullseye install.

commit 1c5e4864 (2022-01-29): Use a{sv} signature for ConnectDevice call.

Fixes #22.