bluez/bluer #24: How do I avoid os: error 11 (connection refused).

StratusFearMe21, 2022-02-16:
Hello, I am attempting to make a lightweight device driver for the LG Magic Remote. However, when I try to connect to it through `l2cap` it errors out with an `os error 111`. This is my code so far. It just tries to connect to the Magic Remote and it begins to receive packets.
```rust
use bluer::{
    l2cap::{SocketAddr, Stream, PSM_BR_EDR_DYN_START, PSM_LE_DYN_START},
    Address,
};
use tokio::io::AsyncReadExt;

#[tokio::main]
async fn main() -> bluer::Result<()> {
    let addr: Address = "7C:02:BC:C5:14:3B".parse().unwrap();
    let socket_addr = SocketAddr::new(addr, bluer::AddressType::LePublic, PSM_LE_DYN_START + 5);
    println!("Connecting to magic remote");
    let mut stream = Stream::connect(socket_addr).await?;
    println!("Local address: {:?}", stream.as_ref().local_addr()?);
    println!("Remote address: {:?}", stream.peer_addr()?);
    println!("Recv MTU: {}", stream.as_ref().recv_mtu()?);
    println!("Security: {:?}", stream.as_ref().security()?);

    loop {
        let mut hello_buf = [0u8; 30];
        stream.read_exact(&mut hello_buf).await?;
        println!("Received: {:?}", hello_buf);
    }
}
```
This is what it prints out
```
Connecting to magic remote
Local address: SocketAddr { addr: E0:D4:E8:38:4D:65, addr_type: LePublic, psm: 133, cid: 64 }
Remote address: SocketAddr { addr: 7C:02:BC:C5:14:3B, addr_type: LePublic, psm: 133, cid: 0 }
Recv MTU: 672
Security: Security { level: Low, key_size: 0 }
Error: Error { kind: Internal(Io(ConnectionRefused)), message: "Connection refused (os error 111)" }
```
Any ideas? Thanks!

StratusFearMe21, 2022-02-16:
P.S there's no evidence of errors caused by app when I run it and then run `sudo systemctl status bluetooth`

surban, 2022-02-16:
1. Are you sure that the LG Magic Remote is accepting L2CAP connections?
2. Are you sure that the port number is correct (`PSM_LE_DYN_START + 5` in your case)?