bluez/bluer #28: [Question] How to send the report update from a server to connected client?

phuongtran7, 2022-03-22:
Hi,

This might be a stupid question as I'm not very familiar with Bluetooth in general. I'm so sorry.

I'm trying to emulate a Bluetooth mouse using the library and a Raspberry Pi. I have implement all the required service and characteristic for HID over GATT. The end device, an iPhone, can see and connect to the Pi running the server. It stays connected and read these characteristic:

```
ManufacturerNameString - Read request CharacteristicReadRequest { offset: 0, mtu: 517, link: Some(Le) }
ReportMap - Read request CharacteristicReadRequest { offset: 0, mtu: 517, link: Some(Le) }
HidInformation - Read request CharacteristicReadRequest { offset: 0, mtu: 517, link: Some(Le) }
BatteryLevel - Read request CharacteristicReadRequest { offset: 0, mtu: 517, link: Some(Le) }
```

However, I'm not sure how I can start sending the update to the iOS device, for example new X/Y coordinate. I have a `Characteristic::Report` that has read and notify implement, but it never gets called for some reason. I also have another `Characteristic::BootMouseInputReport` with read and write, but that is not called either. After connecting to the iOS device, it just stays there.

I looked around a bit and found the example for writing notification using IO, but that still require the iOS device to initiate the notification session first, I think.

So my question is how can I send the update to iOS device after connected? Seems like no way to get the writer without receiving the notification request first.

The services and characteristic I have are:

```
HID Service
    Report
    Protocol Mode
    Report Map
    Boot Mouse Input Report
    Hid Information
    Hid Control Point

Battery Service
    Battery Level

Device Information Service
    Manufacturer Name String
```

Please let me know if this is not the appropriate place for this. I will try and remove it asap.

Thank you so much for your time and help.

phuongtran7, 2022-03-23:
Solved!

It seems that you will need to add `bluer::id::Descriptor::ReportReference` into the `bluer::id::Characteristic::Report` for the notification to get called correctly. I'm now able to report mouse movement to the iOS device.

closed 2022-03-23 by phuongtran7

surban, 2022-03-24:
Generally the client device has to enable notifications, otherwise there is no way for you to send them.

phuongtran7, 2022-03-24:
I see. Thanks. Seems like I didn't implement the spec correctly or iOS expects some extra characteristic or something. But after adding the `Report Reference` characteristic descriptor to my `Input Report`, the client does initiate the notification for the `Report` now and I'm now able to constantly sending the data over.

Thank you so much for the library too. Much nicer than wrestling with BlueZ source code trying to piece together a server.