bluez/bluer #36: 'Invalid argument "mtu"' when attempting to write local characteristic

BroderickCarlin, 2022-05-02:
Attempting to setup a GATT peripheral with a basic writeable characteristic but attempts to actually perform a write (or write without response) consistently result in the following error: 

```
[2022-04-27T22:13:26Z TRACE bluer] /org/bluez/bluer/gatt/app/4bef7b60a5124fc0b5de51b736b207a8/service0/char1: org.bluez.GattCharacteristic1.WriteValue ([1], {"link": Variant("LE"), "device": Variant(Path("/org/bluez/hci0/dev_7C_3E_56_10_96_8E\u{0}"))})
[2022-04-27T22:13:26Z TRACE bluer] /org/bluez/bluer/gatt/app/4bef7b60a5124fc0b5de51b736b207a8/service0/char1: org.bluez.GattCharacteristic1.WriteValue (...) -> Err(MethodErr(ErrorName("org.freedesktop.DBus.Error.InvalidArgs\u{0}"), "Invalid argument \"mtu\""))
```

This same error occurs when running the example GATT servers provided as well as the GATT server(s) hosted by bluer-tools. 

Bluer version: 0.14.0
Bluez version: 5.48
D-Bus Message Bus Daemon: 1.12.2
libdbus-1-dev: 1.12.2-1ubuntu1.2

BroderickCarlin, 2022-05-02:
As a bit of a follow up; it appears the issue originates from this line here: https://github.com/bluez/bluer/blob/567b0fc760ce4fe7f9f10c5ea82e9a031cbeec19/bluer/src/gatt/local.rs#L479

It does not appear that on my system an `mtu` field is being included in the dbus message and thus that prop needed to be marked as optional. Curious though if this is by design or if the change to make the `mtu` optional should be upstreamed.

surban, 2022-05-05:
Could you try with BlueZ 5.60? Versions before that had several bugs related to GATT handling.

closed 2022-05-05 by BroderickCarlin