bluez/bluer #43: Cant build examples

vChavezB, 2022-08-19:
Hi there,

I am quite new to rust and want to test this library but I cant import bluer correctly. I get the following error:

```
error[E0432]: unresolved imports `bluer::Adapter`, `bluer::AdapterEvent`, `bluer::DeviceEvent`
```

My Cargo manifest looks as follows 

```toml
[package]
name = "discover"
version = "0.1.0"
edition = "2021"

[dependencies]
bluer = "0.15.0"
```

and as a source file I am using the [discover example](https://github.com/bluez/bluer/blob/master/bluer/examples/discover_devices.rs).

Is there something else I am missing when building the application?

surban, 2022-08-20:
Please use `bluer = { version = "0.15", features = ["full"] }` in your Cargo.toml file.

vChavezB, 2022-08-21:
Just as a reference this is an example of how the cargo manifest should look like to compile the examples as standalone applications

```
[package]
name = "discover"
version = "0.1.0"
edition = "2021"

[dependencies]
bluer = { version = "0.15", features = ["full"] }
tokio = "1.20.1"
futures = "0.3.23"
env_logger = "0.9.0"

```

closed 2022-08-21 by vChavezB