bluez/bluer #44: Gatt Server reports "Not Registered"

Severon96, 2022-08-25:
Hey guys,

can someone help me please, because i'm a little confused what this message means?
`Error: Error { kind: NotRegistered, message: "" }`

I get this error when starting the GATT-Server implementation that is documented here:
https://docs.rs/bluer/latest/src/gatt_server_io/gatt_server_io.rs.html

The only thing that's changed is that I added some UUIDs for the service(0000ffe0-0000-1000-8000-00805F9B34FB) and characteristics (0000ffe1-0000-1000-8000-00805F9B34FB). 
I guess that I have to register them somewhere but I don't know how or where.

Maybe someone can help me here? Because I don't know what the problem is.

surban, 2022-08-29:
Does the corresponding GATT client example work, when used together with the GATT server example?

What BlueZ version are you using?

Severon96, 2022-08-29:
Nope, I don't get the examples working. They're always failing with this Error:
```
thread 'main' panicked at 'services: Os { code: 2, kind: NotFound, message: "No such file or directory" }', bluer/build.rs:256:6
```

My BlueZ version is: 5.43-2+rpt2+deb9u2

surban, 2022-08-29:
Please make sure you cloned the repository with submodules as described in the README.

Severon96, 2022-08-30:
I've forgot the `--recursive` flag when checking out, that's correct. But nevertheless I added it when checking out again I still receive the "NotFound" error when running the example. But now without any message. 

```
pi@raspberrypi:~/bluer/bluer $ cargo run --all-features --example gatt_server_io   Compiling tokio v1.20.1
   Compiling dbus v0.9.6
   Compiling dbus-crossroads v0.5.1
   Compiling dbus-tokio v0.7.5
   Compiling tokio-stream v0.1.9
   Compiling bluer v0.15.0 (/home/pi/bluer/bluer)
    Finished dev [unoptimized + debuginfo] target(s) in 6m 39s
     Running `/home/pi/bluer/target/debug/examples/gatt_server_io`
Error: Error { kind: NotFound, message: "" }
```

surban, 2022-08-30:
Could you run `bluetoothctl` and verify that an adapter is present?

Severon96, 2022-08-30:
The adapter wasn't present, I fixed that now. But I now get another error when trying to launch the example:

```
pi@raspberrypi:~/bluer $ cargo run --all-features --example gatt_server_io
    Finished dev [unoptimized + debuginfo] target(s) in 1.73s
     Running `target/debug/examples/gatt_server_io`
Error: Error { kind: Internal(DBus("org.freedesktop.DBus.Error.AccessDenied")), message: "Rejected send message, 2 matched rules; type=\"method_call\", sender=\":1.9\" (uid=1000 pid=2927 comm=\"target/debug/examples/gatt_server_io \") interface=\"org.freedesktop.DBus.ObjectManager\" member=\"GetManagedObjects\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.bluez\" (uid=0 pid=459 comm=\"/usr/local/libexec/bluetooth/bluetoothd --experime\")" }
```

I made experience that if I build my code and run the compiled version with sudo this error is fixed. But that doesn't work for the example because it throws an "NotRegistered"-Error for me then.

Severon96, 2022-09-02:
So, sorry that you didn't hear anything from me in the past days. The update of BlueZ didn't work very well on my old image with Raspbian Stretch running on it. 
I've created a new image now and updated to 5.65. 
Now the gatt_server_io example is starting and I can search with some bluetooth-tools for it. Also my code is working, too (not wondering, since it's nearly an exact copy of the example, but Hey. Worth mentioning!). 
Thanks for your help!

surban, 2022-10-01:
Great to hear it's working now!

closed 2022-10-01 by surban