bluez/bluer #47: Second discovery attempt fails with InProgress error

jmagnuson, 2022-09-02:
Device: Raspberry Pi zero 2w
OS: Yocto Linux
BlueZ: 5.64

Running the `gatt_client` example the first time succeeds as expected:

```
root@yocto:/tmp# ./gatt_client
Discovering on Bluetooth adapter hci0 with address B8:27:EB...

...

Discovered device B8:27:EB... with service UUIDs {00000002-e900-4e66-854c-cde416ae1332, 00000000-0000-0000-0000-0000feedc0de}
    Manufacturer data: Some(...)
    Device provides our service!
    Connecting...
    Connect error: Bluetooth operation failed: le-connection-abort-by-local
    Connected
    Enumerating services...
    Service UUID: 00001801-0000-1000-8000-00805f9b34fb
    Service data: [Primary(true), Uuid(00001801-0000-1000-8000-00805f9b34fb), Includes([])]
    Service UUID: 00001800-0000-1000-8000-00805f9b34fb

...

    Stopping notification IO
    Characteristic exercise completed
    Device disconnected

Stopping discovery
```

And I can see both `StartDiscovery` and `StopDiscovery` going over DBus:

```
method call time=1662105582.357526 sender=:1.71 -> destination=org.bluez serial=12 path=/org/bluez/hci0; interface=org.bluez.Adapter1; member=StartDiscovery
method return time=1662105582.361596 sender=:1.61 -> destination=:1.71 serial=189 reply_serial=12
signal time=1662105582.367611 sender=:1.61 -> destination=(null destination) serial=190 path=/org/bluez/hci0; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.Adapter1"
   array [
      dict entry(
         string "Discovering"
         variant             boolean true
      )
   ]
   array [
   ]

...

method call time=1662105596.823707 sender=:1.71 -> destination=org.bluez serial=16 path=/org/bluez/hci0; interface=org.bluez.Adapter1; member=StopDiscovery
method return time=1662105596.828165 sender=:1.61 -> destination=:1.71 serial=222 reply_serial=16
```

However, subsequent attempts fail with `org.bluez.Error.InProgress`:

```
root@yocto:/tmp# ./gatt_client
Discovering on Bluetooth adapter hci0 with address B8:27:EB...

Error: Error { kind: InProgress, message: "Operation already in progress" }

root@yocto:/tmp# bluetoothctl show | grep Discovering
        Discovering: no
```

```
method call time=1662107942.325967 sender=:1.80 -> destination=org.bluez serial=10 path=/org/bluez/hci0; interface=org.bluez.Adapter1; member=StartDiscovery
error time=1662107942.330377 sender=:1.61 -> destination=:1.80 error_name=org.bluez.Error.InProgress reply_serial=10
   string "Operation already in progress"
```

I have been able to correct this by power cycling with `bluetoothctl power on/off`, but ideally there's something that can be done in bluer to avoid this.

I'm not an expert on the BlueZ DBus API, but it seems like it may have to do with the device connection being made after discovery, since I do not see this issue with the `discover_devices` example, which only does device discovery and no connection attempts.

surban, 2022-10-01:
It is difficult to say what is causing it. You will probably have to look into `bluetoothd` why it is rejecting subsequent discovery requests. As a workaround you could reset the adapter's power using BlueR, but of course this is not a good solution.

I am tagging this as external, because according to your description this behavior is caused by the Bluetooth daemon.

surban, 2024-02-01:
I am closing as completed for now, since there has been no response. Feel free to reopen if the issue still exists.

closed 2024-02-01 by surban