bluez/bluer #53: Cannot build on musl targets

lu-zero, 2022-10-18:
While building for `aarch64-unknown-linux-musl`:
```
   Compiling bluer v0.15.1 (/tmp/bluer/bluer)
error[E0308]: mismatched types
   --> bluer/src/sys.rs:188:38
    |
188 | pub const RFCOMMCREATEDEV: c_ulong = request_code_write!('R', 200, size_of::<c_int>());
    |                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `u64`, found `i32`
    |
    = note: this error originates in the macro `ioc` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0308]: mismatched types
   --> bluer/src/sys.rs:189:39
    |
189 | pub const RFCOMMRELEASEDEV: c_ulong = request_code_write!('R', 201, size_of::<c_int>());
    |                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `u64`, found `i32`
    |
    = note: this error originates in the macro `ioc` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0061]: this function takes at least 2 arguments but 3 arguments were supplied
   --> bluer/src/sock.rs:303:24
    |
303 |     let ret = unsafe { libc::ioctl(socket.as_raw_fd(), request, value.as_mut_ptr()) };
    |                        ^^^^^^^^^^^                     -------  ------------------ argument unexpected
    |                                                        |
    |                                                        expected `i32`, found `u64`
    |
note: function defined here
   --> /home/lu_zero/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.135/src/unix/linux_like/linux/musl/mod.rs:727:12
    |
727 |     pub fn ioctl(fd: ::c_int, request: ::c_int, ...) -> ::c_int;
    |            ^^^^^
help: you can convert a `u64` to an `i32` and panic if the converted value doesn't fit
    |
303 |     let ret = unsafe { libc::ioctl(socket.as_raw_fd(), request.try_into().unwrap(), value.as_mut_ptr()) };
    |                                                               ++++++++++++++++++++
help: remove the extra argument
    |
303 |     let ret = unsafe { libc::ioctl(socket.as_raw_fd(), /* i32 */) };
    |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

error[E0061]: this function takes at least 2 arguments but 3 arguments were supplied
   --> bluer/src/sock.rs:314:24
    |
314 |     let ret = unsafe { libc::ioctl(socket.as_raw_fd(), request, value as *const _) };
    |                        ^^^^^^^^^^^                     -------  ----------------- argument unexpected
    |                                                        |
    |                                                        expected `i32`, found `u64`
    |
note: function defined here
   --> /home/lu_zero/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.135/src/unix/linux_like/linux/musl/mod.rs:727:12
    |
727 |     pub fn ioctl(fd: ::c_int, request: ::c_int, ...) -> ::c_int;
    |            ^^^^^
help: you can convert a `u64` to an `i32` and panic if the converted value doesn't fit
    |
314 |     let ret = unsafe { libc::ioctl(socket.as_raw_fd(), request.try_into().unwrap(), value as *const _) };
    |                                                               ++++++++++++++++++++
help: remove the extra argument
    |
314 |     let ret = unsafe { libc::ioctl(socket.as_raw_fd(), /* i32 */) };
    |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

error[E0308]: mismatched types
   --> bluer/src/l2cap.rs:404:64
    |
404 |         let value: c_int = sock::ioctl_read(self.fd.get_ref(), TIOCINQ)?;
    |                            ----------------                    ^^^^^^^ expected `u64`, found `i32`
    |                            |
    |                            arguments to this function are incorrect
    |
note: function defined here
   --> bluer/src/sock.rs:301:8
    |
301 | pub fn ioctl_read<T>(socket: &OwnedFd, request: c_ulong) -> Result<T> {
    |        ^^^^^^^^^^    ----------------  ----------------
help: you can convert an `i32` to a `u64` and panic if the converted value doesn't fit
    |
404 |         let value: c_int = sock::ioctl_read(self.fd.get_ref(), TIOCINQ.try_into().unwrap())?;
    |                                                                       ++++++++++++++++++++

error[E0308]: mismatched types
   --> bluer/src/l2cap.rs:412:64
    |
412 |         let value: c_int = sock::ioctl_read(self.fd.get_ref(), TIOCOUTQ)?;
    |                            ----------------                    ^^^^^^^^ expected `u64`, found `i32`
    |                            |
    |                            arguments to this function are incorrect
    |
note: function defined here
   --> bluer/src/sock.rs:301:8
    |
301 | pub fn ioctl_read<T>(socket: &OwnedFd, request: c_ulong) -> Result<T> {
    |        ^^^^^^^^^^    ----------------  ----------------
help: you can convert an `i32` to a `u64` and panic if the converted value doesn't fit
    |
412 |         let value: c_int = sock::ioctl_read(self.fd.get_ref(), TIOCOUTQ.try_into().unwrap())?;
    |                                                                        ++++++++++++++++++++

error[E0308]: mismatched types
   --> bluer/src/rfcomm/mod.rs:254:64
    |
254 |         let value: c_int = sock::ioctl_read(self.fd.get_ref(), TIOCINQ)?;
    |                            ----------------                    ^^^^^^^ expected `u64`, found `i32`
    |                            |
    |                            arguments to this function are incorrect
    |
note: function defined here
   --> bluer/src/sock.rs:301:8
    |
301 | pub fn ioctl_read<T>(socket: &OwnedFd, request: c_ulong) -> Result<T> {
    |        ^^^^^^^^^^    ----------------  ----------------
help: you can convert an `i32` to a `u64` and panic if the converted value doesn't fit
    |
254 |         let value: c_int = sock::ioctl_read(self.fd.get_ref(), TIOCINQ.try_into().unwrap())?;
    |                                                                       ++++++++++++++++++++

error[E0308]: mismatched types
   --> bluer/src/rfcomm/mod.rs:262:64
    |
262 |         let value: c_int = sock::ioctl_read(self.fd.get_ref(), TIOCOUTQ)?;
    |                            ----------------                    ^^^^^^^^ expected `u64`, found `i32`
    |                            |
    |                            arguments to this function are incorrect
    |
note: function defined here
   --> bluer/src/sock.rs:301:8
    |
301 | pub fn ioctl_read<T>(socket: &OwnedFd, request: c_ulong) -> Result<T> {
    |        ^^^^^^^^^^    ----------------  ----------------
help: you can convert an `i32` to a `u64` and panic if the converted value doesn't fit
    |
262 |         let value: c_int = sock::ioctl_read(self.fd.get_ref(), TIOCOUTQ.try_into().unwrap())?;
    |
```

I wonder if the socket abstraction couldn't now go away since std has its own OwnedFd now.

surban, 2022-10-18:
Hi!

Thanks for reporting this.

Would you be willing to submit a PR that includes `musl` targets in CI testing?

lu-zero, 2022-10-18:
It would be probably easier once bluer moves to zbus though, spinning up an alpine image is the easiest way to test otherwise. At least that I can think of.

surban, 2022-10-19:
Yes, moving to zbus would indeed be great. The only limiting factor here is manpower and that libdbus works well enough currently.

closed 2022-10-19 by surban