bluez/bluer #62: Multi-central Nordic UART Service implementation

alexandrefresnais, 2022-12-14:
Hello,

First, thanks for the amazing work!

I would like to know if it should be possible to implement the Nordic UART Service specification with this API ?

I know this question looks related to https://github.com/bluez/bluer/issues/58 but I cannot precisely see how it could be done in this specific case.

The Nordic documentation specifies that a central client may write on a RX Characteristic, and get answered with a notification.
As an example, I took a look at the `gatt_echo_server` and I could not see how I could echo the specific client which wrote me.

I understand that I can have multiple Write and Notify events, but I cannot see how I could link a writer and a notifier to answer the good client.

Is it possible ?

Thanks!

Regards,

Alexandre

surban, 2022-12-14:
> I understand that I can have multiple Write and Notify events, but I cannot see how I could link a writer and a notifier to answer the good client.

I see your point. You would need to know the address of the client device to link the write and notify events.

It seems that the client device is indeed provided via the `device` property over D-Bus, see https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt#n105 for example, but BlueR is currently not exposing that property.

surban, 2022-12-14:
Note that you will have to use IO-based notifications, because BlueZ does not provide the requesting device when using callback-based notifications. See https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt#n185

closed 2022-12-14 by surban

surban, 2022-12-14:
> Wow, thank you very much. I will surely be taking a look at this with an iPhone in the following weeks. Would you be interested in a new example if I were making one ?

Yes, definitely!

commit 85c93de4 (2022-12-14): Expose device making request in GATT server.

Provide access to name of adapter and address of device that is
responsible for a GATT server request.

Fixes #62