bluez/bluer #66: Problem importing bluer

goodpilotfish, 2023-01-26:
Hi, I'm having some problems using bluer. I assume there is a problem with my Cargo.toml file but I don't see my error.

Error:
```
   Compiling bluer v0.15.5
   Compiling ble v0.1.0 (/home/sarabrodin/playground/mossmoss/ble)
error[E0432]: unresolved imports `bluer::AdapterEvent`, `bluer::DeviceEvent`
 --> src/main.rs:1:13
  |
1 | use bluer::{AdapterEvent, DeviceEvent};
  |             ^^^^^^^^^^^^  ^^^^^^^^^^^ no `DeviceEvent` in the root
  |             |
  |             no `AdapterEvent` in the root
```
```
error[E0433]: failed to resolve: could not find `Session` in `bluer`
 --> src/main.rs:6:26
  |
6 |     let session = bluer::Session::new().await?;
  |                          ^^^^^^^ could not find `Session` in `bluer`
```

Cargo:
```
[package]
name = "ble"
version = "0.1.0"
edition = "2021"

[dependencies]
bluer = "0.15"
tokio = { version = "1.24", features = ["full"] }
futures = "0.3"
```

Code:
Similar to the discover device example.

Thanks!

goodpilotfish, 2023-01-26:
Note:
```
[xxx@fedora ble]$ rustc --version
rustc 1.66.1 (90743e729 2023-01-10) (Fedora 1.66.1-1.fc36)
```

When I clone the bluer repo I can run the examples on the same mashine.

EduardoPD1921, 2023-01-27:
I'm facing the exactly same problem. Did you find some fix for this?