bluez/bluer #90: Is this supposed to be private?

alxpettit, 2023-07-24:
I can't use the AsyncRead or AsyncWrite traits for Stream, defined in `src/rfcomm/mod.rs`. The compiler says they're private. But how else am I supposed to write to a stream?

surban, 2023-07-25:
There is no such thing as a private implementation of a public trait in Rust. `AsyncRead` and `AsyncWrite` are public, as you can see from the [docs](https://docs.rs/bluer/latest/bluer/rfcomm/struct.Stream.html#impl-AsyncRead-for-Stream).

jmagnuson, 2023-08-06:
@alxpettit what is the error message from the compiler? Is the `rfcomm` [feature](https://github.com/bluez/bluer#crate-features) enabled?

surban, 2023-08-19:
Closing, because no further information was provided. Feel free to re-open if the issue is not solved.

closed 2023-08-19 by surban