bluez/bluer #105: `Adapter::discover_devices` fails with bluez 5.50: "Invalid arguments in method call"

not-my-profile, 2023-09-16:
```rs
#[tokio::main]
async fn main() {
    let session = bluer::Session::new().await.unwrap();
    let adapter = session.default_adapter().await.unwrap();
    adapter.set_powered(true).await.unwrap();

    let events = adapter
        .discover_devices()
        .await
        .expect("discover devices failed");
}
```

Failed with the following message on my laptop:

```
thread 'main' panicked at 'discover devices failed: Error { kind: InvalidArguments, message: "Invalid arguments in method call" }', src/main.rs:10:10
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

which had bluez 5.50-1.2~deb10u3 (Debian buster) installed.

Upgrading to Debian bullseye (bluez 5.55-3.1) fixed the problem.

I don't know if this library is intended to support bluez 5.50, however I think in any case it probably makes sense for `Session::new` to perform a version check, returning an error if the detected bluez version is lower than the minimum supported bluez version by this library.

surban, 2023-09-19:
I agree that this would be helpful, but unfortunately `bluetoothd` does not expose its version number over D-Bus, thus we cannot perform a version check.

not-my-profile, 2023-09-19:
Ah yeah ... that is unfortunate. Checking the output of `/usr/sbin/bluetoothd --version` would of course not be completely reliable since there is no guarantee that this is actually the daemon running (which I think precludes it from being done automatically on `Session::new`). Perhaps a new `assert_bluetoothd_version_is_supported` method could be introduced that checks `/usr/sbin/bluetoothd --version` (with the caveat that this might not be the bluetoothd version actually running being documented)?

surban, 2023-10-06:
This would be a possibility. However, I am a bit reluctant to add a feature that is potentially unreliable and maybe misleading due to the limitations you described. 

I believe the best option is to ask the BlueZ developers to expose their version number over D-Bus and see what they think about it.

surban, 2023-10-06:
Feature request opened at at https://github.com/bluez/bluez/issues/619.

not-my-profile, 2023-10-06:
Thanks for opening the feature request there! I gave it a :+1:.

The only caveat I see is that the new feature probably won't be backported, so it will probably take quite some time till the bluez version in Debian stable has this feature ... but I'm looking forward to a future where you could rely on this .. even if it's distant.