bluez/bluer #106: `connect_device` fails with D-Bus error org.freedesktop.DBus.Error.UnknownMethod

not-my-profile, 2023-09-16:
I tried calling `Adapter::connect_device` and it failed with:

> internal error: D-Bus error org.freedesktop.DBus.Error.UnknownMethod: Method "ConnectDevice" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist

I know that the documentation says:

> This method is experimental.

However this seems to be a different issue since it fails immediately? Perhaps my bluez version doesn't support it? This was encountered with bluez 5.55-3.1 (debian).

not-my-profile, 2023-09-16:
Ah ok my bad ... apparently `bluetoothd` has to be run with `--experimental` (or `Experimental = true` needs to be set in `/etc/bluetooth/main.conf`, although this didn't work for me on Debian 11) in order for the ConnectDevice method to be available.

closed 2023-09-16 by not-my-profile

LeCyberDucky, 2024-06-29:
Sorry for reviving this issue, but what do you mean by running `bluetoothd` with `--experimental`? 
Is that something you do in the code?