bluez/bluer #111: Is setting the L2CAP LE LL PDU possible? If so, how would this be done in BlueR?

Jah-On, 2023-11-21:
Hello!

I am working on a project trying to implement [Google's ASHA](https://source.android.com/docs/core/connect/bluetooth/asha) on Linux. The problem that I seem to be running into is not being able to [set the PDU data as needed](https://source.android.com/docs/core/connect/bluetooth/asha#audio-packet-format-and-timing). This may be due to an inadequate amount of knowledge I have about Bluetooth and BlueZ/R. I am new to it. 

I have been using Stream as that's what I believe the processors are using but I've also tried SeqPacket. Here are my questions..
- When writing data to Stream, are we responsible for PDU, L2CAP header and SDU?
- If BlueZ/R automatically manages those parts, is there a way they can be modified after writing the data but before flushing?

Thank you for taking the time to read this!

Vudentz, 2023-11-21:
@Jah-On assuming BlueR uses L2CAP sockets then only the data itself needs to be transferred, all the L2CAP signalling will be handled automatically, including, headers, segmentation, credits, etc.

Jah-On, 2023-11-21:
@Vudentz Thank you for your reply. Here's an image that I am using as a reference...

[image]

As listed in the ASHA documentation above, it is crucial for the streaming to work that I modify the two bytes of the PDU. Is there a way I can modify the PDU bytes in BlueZ?  Maybe in either the socket options or in the file descriptor?

Any insight would be much appreciated!

shermp, 2023-11-21:
@Jah-On I'm no bluetooth expert (but I'm eagerly watching LASHA), but is Data Length Extensions (DLE) enabled?

shermp, 2023-11-21:
Also, just throwing ideas out there, does MTU need to be specified?

Jah-On, 2023-11-21:
> Also, just throwing ideas out there, does MTU need to be specified?

No as this is set by the peripheral. At 20ms intervals (160 bytes), I am well under the L2CAP LE data frame MTU of 251 bytes.

Jah-On, 2023-11-21:
> @Jah-On I'm no bluetooth expert (but I'm eagerly watching LASHA), but is Data Length Extensions (DLE) enabled?

I haven't tried this yet, I don't think... can't remember lol. However, I don't think this gets at the underlying issue of needing to modify the PDU. I'll keep it in mind tho.

Vudentz, 2023-11-21:
> @Vudentz Thank you for your reply. Here's an image that I am using as a reference...
>
> [image]
>
> As listed in the ASHA documentation above, it is crucial for the streaming to work that I modify the [...]

Where exactly are these 2 bytes that you want to change, L2CAP header? Not sure why it is even mentioned on the ASHA documentation about the layer bellow HCI, afaik even with the likes of Android you can't really do anything about that, as for L2CAP the headers are also subject to qualification so Id be really surprised if they don't comply with the standard.

The actually L2CAP PDU format is the following:

[image]

Even the SDU header is actually appended based on the length of the information payload, perhaps the idea is to avoid L2CAP segmentation, HCI fragmentation and LL segmentation altogether which means the MTU needs to be the smallest of the 3, in that case you can probably set the BT_MTU to 0 which when then make it automatically select whatever was set in the hci_conn->mtu:

https://github.com/bluez/bluetooth-next/commit/4b6e228e297b73451f3a4b12fb7d0b24d9d32e6f

shermp, 2023-11-21:
Looking at the source code for ASHA in Flouride (the Android bluetooth stack), it seems it makes a call to `BTM_SetBleDataLength()`, which appears to be setting the link layer PDU size?

Jah-On, 2023-11-22:
> > @Vudentz Thank you for your reply. Here's an image that I am using as a reference...
> > [image]
> > As listed in the ASHA documentation above, it is crucial for the streaming to work that I modify the two bytes of the PDU. Is there a way I can modify the PDU bytes in BlueZ? Maybe in either the socket options or in the file descriptor?
> 
> Where exactly are these 2 bytes that you want to change, L2CAP header? Not sure why it is even mentioned on the ASHA documentation about the layer bellow HCI, afaik even with the likes of Android you can't really do anything about that, as for L2CAP the headers are also subject to qualification so Id be really surprised if they don't comply with the standard.
> 
> The actual L2CAP PDU format is the following:
> 
> [image]
> 
> Even the SDU header is actually appended based on the length of the information payload, perhaps the idea is to avoid L2CAP segmentation, HCI fragmentation and LL segmentation altogether which means the MTU needs to be the smallest of the 3, in that case you can probably set the BT_MTU to 0 which when then make it automatically select whatever was set in the hci_conn->mtu:
> 
> [bluez/bluetooth-next@4b6e228](https://github.com/bluez/bluetooth-next/commit/4b6e228e297b73451f3a4b12fb7d0b24d9d32e6f)

Here is the Android source for the hearing aid streaming...[https://android.googlesource.com/platform/packages/modules/Bluetooth/+/refs/heads/main/system/bta/hearing_aid/hearing_aid.cc](https://android.googlesource.com/platform/packages/modules/Bluetooth/+/refs/heads/main/system/bta/hearing_aid/hearing_aid.cc

From what I can gather, they are generating their own packet and sending it via GAP but I may be wrong on that. If you could read their source code and say whether getting same result is possible with BlueZ/R, that would be much appreciated. You are much more informed in this area and I feel overwhelmed after trying to understand what I should be doing for two weeks. 

It is entirely possible that this goes beyond the intended scope of BlueZ but I would rather hear that coming from you than falsely come to that conclusion myself. Thanks for taking the time to discuss this.

surban, 2023-11-22:
BlueR exposes all available L2CAP socket options via [`Socket<_>`](https://github.com/bluez/bluer/blob/5fd69334a1ec76e56953b3727d63303302da8942/bluer/src/l2cap.rs#L231). So far I am not aware of any socket option to explicitly set the PDU. This would first have to be implemented in the Linux kernel, before we can expose it in BlueR.

Jah-On, 2023-11-22:
I took another look at the Android source code and I now think the sequence counter is actually in the data part of the packet. Thus, I shouldn't need to modify the generated packet components. However, I have tried this as well as two other developers for their MacOS implementation and the streaming isn't behaving as expected. When we send the packet data into BlueZ/R, is it expecting the data to arrive in a specific endian/orientation?

surban, 2023-11-22:
I am not aware of any endian-requirements.

Could you verify that L2CAP sockets work for you between two Linux hosts using `l2cat`?

Jah-On, 2023-11-22:
> I am not aware of any endian-requirements.
> 
> Could you verify that L2CAP sockets work for you between two Linux hosts using `l2cat`?

No as the device(s) we are interfacing with are hearing aids or cochlear implant processors that only run embedded closed source firmware. Perhaps I could see if any open source RTOS projects have implemented ASHA but I doubt that will yield anything.

Jah-On, 2023-11-27:
As it turns out, this is not needed. However, we do need a way to set the connection interval to 20ms. It looks like this is already implemented for BlueZ but is not a part of BlueR yet. Can this be added and should I rename this "issue" or close it and make a new one?

surban, 2023-11-28:
Please create a new issue.

closed 2023-11-28 by Jah-On