bluez/bluer #112: Can setting the connection interval parameter be added?

Jah-On, 2023-11-28:
Hello,

Is it possible to set the connection interval in BlueR or is this a "wait for bluetoothd to add it then we will" issue? 

Thanks

surban, 2023-11-28:
What method do you use to set the connection interval?

Jah-On, 2023-11-28:
> What method do you use to set the connection interval?

We think it's  LoadConnectionParameters but we have not tried it with a command line utility yet. We try 

`
echo "16" > /sys/kernel/debug/bluetooth/hci0/conn_min_interval
echo "16" > /sys/kernel/debug/bluetooth/hci0/conn_max_interval
`

as sudo but could not modify the file even after stopping BlueZ.

surban, 2023-11-28:
Given the path it seems to be a debug-only setting.

@Vudentz Do you know a supported method to achieve the same result? Does it make sense to extend the Bluetooth D-Bus API to change adapter settings like these?

Jah-On, 2023-11-28:
Without this ability, ASHA can not be implemented for Linux. As we understand it, the connection interval must be set to 10 or 20 ms for hearing aids and cochlear implants to properly receive the data. For some reason, Cochlear's N7 advertises a max interval of 30ms but this will not work for streaming. Google's ASHA implementation forces the interval to be either 10 or 20ms.

hxk1633, 2023-11-28:
I support adding this enhancement. Setting the connection interval is required implement ASHA on Linux.

Jah-On, 2024-01-22:
Hello,

Just checking in to see if there are any updates on if or when this will be added. I reckon I could use the old BlueZ 4 kernel API to set the connection interval. This possibly can be done with FFI but I'm not sure if that's a good idea. Ideally, everything should be managed via BlueR.

Thanks!

OmarCastro, 2024-01-22:
Hello,

I am also interested in ASHA in linux, while I do not undestand much about the internals of blueZ, there is this code about the HCI, https://github.com/bluez/bluez/blob/master/lib/hci.c , that helps in defining the connection interval

I think creating Rust bindings to the HCI part of blueZ, is a good start to solve this issue, what do y'all think?

surban, 2024-01-25:
Although I like the idea of adding a HCI interface to BlueR, I don't think that it is the right solution to this particular problem. You will run into permission problems, since HCI functions are not available to non-root users. Furthermore, to my knowledge `bluetoothd` (if running) should be the only process using the HCI interface.

While I am not an expert on this particular issue, I suppose the way forward is to implement functionality to set the connection interval in `bluetoothd` and expose it via D-Bus. Since it is a global parameter of the Bluetooth controller, `bluetoothd` will have to merge requests from different clients. 

Afterwards it is a simple step to include an API for that in BlueR.

Jah-On, 2024-01-25:
Hey @surban, thanks for your reply! I opened an issue on the main BlueZ repo, https://github.com/bluez/bluez/issues/731, so hopefully we can get somewhere with it. 

Cheers