bluez/bluer #118: Add ability to save LE Advertising Reports with bluer-tools

potto216, 2023-12-26:
It would be useful for one of the bluer-tools to have a command line option to save advertising reports for future analysis and comparisons. Currently I modified `blumon` to save the reports in a JSON array with an example given below:

```
[{
  "local_name": "",
  "address": "61:2E:6D:11:B5:82",
  "address_type": "random",
  "manufacturer_data": "",
  "service_data": "",
  "last_seen": 0,
  "RSSI": -76
},
{
  "local_name": "",
  "address": "56:04:B8:80:A5:AE",
  "address_type": "random",
  "manufacturer_data": "",
  "service_data": "",
  "last_seen": 0,
  "RSSI": -71
},
```

Is there interest in accepting a PR for either modifying an existing bluer-tools program or creating a new one to include this functionality? I think having bluer-tools that can run at the command line by users not familiar with Rust will increase the adoption of BlueR and fill a gap in BlueZ command line tool capability, especially since the hci tools are deprecated.

The requirements of the feature would be:
1. Provide a command line feature to save received advertising reports. The minimum information to be included is the information from either the LE Advertising Report event or the LE Extended Advertising Report event.
2. Provide a command line feature to either write the file as a JSON array or a CSV file without a column header
3. Provide a command line feature to have extra information included with each advertising report such as the scanning adapter public address and the time the report was recorded. 

@surban what are your thoughts?

surban, 2024-02-01:
Yes, sounds good to me!

potto216, 2024-02-03:
Great I'll fork the repo and make the changes on master unless the changes should be in a branch

potto216, 2024-02-11:
@surban [here is a prototype](https://github.com/potto216/bluer/blob/118-Save-LE-Advertising-Reports/bluer-tools/src/blumon.rs) to save advertisements to a file. The prototype does not yet have the command line switches. The basic idea is to call the [append method](https://github.com/potto216/bluer/blob/01a2cd055eae37dfef09b1716ce760c0085c879f/bluer-tools/src/blumon.rs#L69) to add a new json record of the advertisement when it is received. Recorded advertisments look like:
```
{
  "adv_name": {
    "RSSI": -47,
    "address": "5D:2D:65:3D:3B:66",
    "address_type": "random",
    "last_seen": 8,
    "local_name": "",
    "manufacturer_data": "0x004C: [0x16 0x08 0x00 0x54(T) 0xDD 0x5A(Z) 0xB5 0x9E 0xC2 0xE2]",
    "service_data": ""
  },
  "time_recorded": "2024-02-09T19:09:21.337408427+00:00"
},
{
  "adv_name": {
    "RSSI": -34,
    "address": "00:15:A3:00:3E:D3",
    "address_type": "public",
    "last_seen": 3,
    "local_name": "UM34C",
    "manufacturer_data": "0x050E: [0xD3 0x3E(>) 0x00 0xA3 0x15 0x00]",
    "service_data": ""
  },
  "time_recorded": "2024-02-09T19:09:21.353514634+00:00"
},
```
A couple issues to resolve.
1. Is a JSON array the best choice? The common way to exit `bluemon` is with CTRL-C the program currently ends without writing the closing array bracket `]`. Possible solutions I thought of around this are:
   * Don't use JSON, use a format that doesn't require end parameters such as CSV.
   * Always write the closing bracket and then before writing the next advertisement rewind the file pointer to erase that. 
   * Add a handle to catch the CTRL-C event and then write out the `]`. Also in this case the logger could write out the entire file or large chunks of the file so it doesn't slow down the I/O by flushing each advertisement to disk.   
2. Is the placement of the `logger.append` statement in the event loop appropriate? Is there a better design pattern to use.

Once the code changes are done I'll merge this branch into my master for the PR.

potto216, 2024-03-09:
Created PR https://github.com/bluez/bluer/pull/133 for review

potto216, 2024-03-26:
Merge complete, the issue can be closed.

closed 2024-03-26 by potto216