bluez/bluer #126: License

druppy, 2024-02-14:
I have tried to understand what kind of license BlueR has, and I am a little confused.

The LICENSE file seems to contain 3 licenses, where the last one seems like GPL 2, but in crates.io BlueR are licensed as BSD-2-Clause. 

What license are relevant where ?

surban, 2024-02-15:
All code is licensed under the BSD-2-Clause. 

Doc-comments have been partially copied and adapted from https://github.com/bluez/bluez/tree/master/doc which is in a GPL-2 / LGPL-2 licensed repository.

druppy, 2024-02-15:
> All code is licensed under the BSD-2-Clause.
> 
> Doc-comments have been partially copied and adapted from https://github.com/bluez/bluez/tree/master/doc which is in a GPL-2 / LGPL-2 licensed repository.

Thanks, for the clarification ! 

Is the bluez licensing relevant if bluer is using it via dbus, or should the license be replaces with a pure BSD-2-Clause as you (and crates.io) stats ?

surban, 2024-02-15:
It's not about D-Bus usage. It's about the doc-comments that have been copied and adapted from the BlueZ documentation. Since I am now lawyer, I cannot tell you about the legal implications.

If you want to be on the safe side and help to improve the situation, you could submit a PR that rewords the doc-comments.

druppy, 2024-02-18:
> It's not about D-Bus usage. It's about the doc-comments that have been copied and adapted from the BlueZ D-Bus documentation. Since I am now lawyer, I cannot tell you about the legal implications.
> 
> If you want to be on the safe side and help to improve the situation, you could submit a PR that rewords the doc-comments.

Thanks, sounds like a good idea when I understand this license thing a bit better :-)