bluez/bluer #131: gatt_echo example: does not work

vricosti, 2024-03-03:
Hi,
I am currently testing the current master of bluer and starting with the "simple" echo example with the following setup:

**Desktop-PC - Ubuntu 22.04)  - gatt_echo_server:**
- bluer (git rev-parse HEAD: e83517fb1704dfcd0ed45fbb796e002c4dad7fce)
- rustc (1.76.0)
- dbus (1.12.20-2ubuntu4.1)
- bluez (5.64-0ubuntu1.1)

```
vricosti@vricosti-Z490-VISION-D:~/Dev/bluer$ ./target/debug/examples/gatt_echo_server
Advertising on Bluetooth adapter hci0 with address 40:EC:99:0E:C4:7C
Serving GATT echo service on Bluetooth adapter hci0
Echo service ready. Press enter to quit.
Accepting notify request event with MTU 512
```

**Raspberrypi - Debian GNU/Linux 11 (bullseye) - gatt_echo_client:**
- bluer (git rev-parse HEAD: e83517fb1704dfcd0ed45fbb796e002c4dad7fce)
- rustc (1.76.0)
- dbus (1.12.28-0+deb11u1)
- bluez (5.55-3.1+rpt2+deb11u1)

```
pi@raspberrypi:~/Dev/bluer $ ./target/debug/examples/gatt_echo_client
Discovering on Bluetooth adapter hci0 with address DC:A6:32:B8:68:7E
Discovered device 40:EC:99:0E:C4:7C with service UUIDs {0000180a-0000-1000-8000-00805f9b34fb, 00001800-0000-1000-8000-00805f9b34fb, 00001801-0000-1000-8000-00805f9b34fb, 00000000-0000-0000-000f-eedc0de00002}
    Manufacturer data: None
    Device provides our service!
    Already connected
    Enumerating services...
    Service UUID: 00001801-0000-1000-8000-00805f9b34fb
    Service UUID: 0000180a-0000-1000-8000-00805f9b34fb
    Service UUID: 00000000-0000-0000-000f-eedc0de00002
    Found our service!
    Characteristic UUID: 00000000-0000-0000-000f-00dc0de00002
    Found our characteristic!
    Obtained write IO with MTU 512 bytes
    Obtained notification IO with MTU 517 bytes
    Test iteration 0 with data size 512
    Waiting for echo
```
and then nothing happens.

I have also tried to run server on rpi and client on pc and this time it's a bit better:

```
pi@raspberrypi:~/Dev/bluer $ ./target/debug/examples/gatt_echo_server
Advertising on Bluetooth adapter hci0 with address DC:A6:32:B8:68:7E
Serving GATT echo service on Bluetooth adapter hci0
Echo service ready. Press enter to quit.
Accepting notify request event with MTU 512
Accepting write request event with MTU 517
Echoing 512 bytes: [c0, 81, 95, 98] ... [5e, f9, f6, 76]
Echoing 20 bytes: [9c, ce, 5c, 45] ... [4d, ec, c9, e7]

Echoing 512 bytes: [eb, 4a, 96, 3d] ... [81, c3, ff, 75]
Echoing 512 bytes: [57, e2, f9, 96] ... [5a, 58, 3d, 6f]
Echoing 512 bytes: [88, e9, fb, 76] ... [e8, 3b, 97, d]
Read stream ended
```

```
vricosti@vricosti-Z490-VISION-D:~/Dev/bluer$ export RUST_BACKTRACE=full
vricosti@vricosti-Z490-VISION-D:~/Dev/bluer$ ./target/debug/examples/gatt_echo_client
Discovering on Bluetooth adapter hci0 with address 40:EC:99:0E:C4:7C

Discovered device DC:A6:32:B8:68:7E with service UUIDs {00000000-0000-0000-000f-eedc0de00002}
    Manufacturer data: None
    Device provides our service!
[... 90 lines cut ...]
                               at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panicking.rs:552:40
  35:     0x564593061f61 - std::panicking::try::h8bbf75149211aaaa
                               at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panicking.rs:516:19
  36:     0x564593061f61 - std::panic::catch_unwind::h8c78ec68ebea34cb
                               at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panic.rs:142:14
  37:     0x564593061f61 - std::rt::lang_start_internal::{{closure}}::hffdf44a19fd9e220
                               at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/rt.rs:148:48
  38:     0x564593061f61 - std::panicking::try::do_call::hcb3194972c74716d
                               at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panicking.rs:552:40
  39:     0x564593061f61 - std::panicking::try::hcdc6892c5f0dba4c
                               at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panicking.rs:516:19
  40:     0x564593061f61 - std::panic::catch_unwind::h4910beb4573f4776
                               at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panic.rs:142:14
  41:     0x564593061f61 - std::rt::lang_start_internal::h6939038e2873596b
                               at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/rt.rs:148:20
  42:     0x56459294c5ca - std::rt::lang_start::hd5b8d7bf12173c4d
                               at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/rt.rs:165:17
  43:     0x5645928f346e - main
  44:     0x7f8aede29d90 - __libc_start_call_main
                               at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
  45:     0x7f8aede29e40 - __libc_start_main_impl
                               at ./csu/../csu/libc-start.c:392:3
  46:     0x5645928834f5 - _start
  47:                0x0 - <unknown>
```

surban, 2024-03-14:
I can reproduce this using BlueZ 5.68. Older versions of BlueR have exactly the same problem with this version of BlueZ, thus the issue is probably caused by BlueZ.

Could you test with earlier versions of BlueZ and determine the first version that has this issue? I.e. run `git bisect` on the BlueZ source?

surban, 2024-03-14:
BlueZ version 5.72 seems to run better, but I still get some hangs.

wayeast, 2024-06-16:
Have there been any learnings on this issue? I'm also getting hangs and would love to know if there's a way around it (?).

(client running on arch pc)
bluer: main branch `5e34d3e840`
rustc: rustc 1.79.0 (129f3b996 2024-06-10)
dbus: 1.14.10-2
bluez: 5.76-1

(server on rpi5)
bluer: main branch `5e34d3e840`
rustc: rustc 1.79.0 (129f3b996 2024-06-10)
dbus: 1.14.10-1~deb12u1 arm64
bluez: 5.66-1+rpt1+deb12u1 arm64

surban, 2024-06-17:
Could you try with Linux 6.9.5?

CosminPerRam, 2024-07-20:
I too had various problems using GATT because of bluez (was on 5.66), but sticking with the latest available version solved everything (maybe because of the GATT fixes on 67/70/72/74 but didnt had the time to troubleshoot around).

surban, 2024-07-20:
Since it seems to be resolved, I am closing this issue.

closed 2024-07-20 by surban