bluez/bluer #137: Unexpected Pairing Request Dialog When Connecting to GATT Server with BlueR

MaxTheTurtle0, 2024-03-24:
### Problem Description:

When connecting to a GATT server implemented in Rust using this crate, an unexpected pairing request dialog appears on the client device. This differs from my experience using the Node.js bleno package, where no pairing dialog is prompted under similar conditions.

**Message**
[image]

### Expected Behavior:

I would expect to connect to the GATT server without being prompted for pairing, especially since my application does not require a high level of security, and this behavior was not observed when using bleno.

### Current Behavior:

Whenever I attempt to connect to the GATT server from the LightBlue app (or any other BLE scanning/connecting tool), a pairing request dialog appears, asking to confirm a pairing code. This request is not desired and seems to be triggered by some default or underlying security settings in the bluer crate.

### Steps to Reproduce:

Set up a GATT server similar to the gatt_server_io.rs example using the bluer crate on a Raspberry Pi.
Advertise a simple service.
Attempt to connect to this service using the LightBlue app or a similar BLE tool.
Observe the pairing request dialog appearing unexpectedly.

### Environment:

Rust version: 1.77.0
bluer crate version: 0.17.1
Client device and OS version: iPhone XR, iOS 17.4

Are there specific configurations or security settings in bluer that I need to adjust to prevent this pairing dialog from appearing?
Could this issue be related to how the GATT server advertises itself or the security level it requests for connections?

I appreciate any guidance or assistance you can provide to help resolve this issue.

MaxTheTurtle0, 2024-03-25:
> Could you try registering an [`Agent`](https://docs.rs/bluer/latest/bluer/agent/struct.Agent.html) that has all methods set to `None`?

I just tried it but I still get the same message. I used the example code provided in [le_advertise.rs](https://github.com/bluez/bluer/blob/master/bluer/examples/le_advertise.rs) and added the agent like this:

[image]

MaxTheTurtle0, 2024-03-26:
Is there anything else I could try?

surban, 2024-03-27:
If your code is setup correctly, there is probably not much we can do on the BlueR front. BlueR is a frontend to `bluetoothd` which houses the actual GATT implementation and is responsible for things like security.

I recommend asking on the BlueZ slack, what the exact requirements are so that no pairing is requested.

MaxTheTurtle0, 2024-03-27:
Issue resolved: After conducting tests on Android and encountering no issues and with your hint that BlueZ is probably causing this, I was able to narrow down the problem's scope. This led me to a solution found on a [Stack Overflow post](https://stackoverflow.com/questions/54222740/bluez-pairing-issue-with-ios). Executing the following commands resolved the issue:

```bash
btmgmt power off
btmgmt bredr off
btmgmt bondable off
btmgmt io-cap 0
btmgmt power on
```

Thank you for your patience and assistance. Apologies for any inconvenience caused.

closed 2024-03-27 by MaxTheTurtle0

surban, 2024-03-28:
Could you check whether using [`set_pairable`](https://docs.rs/bluer/latest/bluer/struct.Adapter.html#method.set_pairable) works as well?

MaxTheTurtle0, 2024-04-03:
> Could you check whether using [`set_pairable`](https://docs.rs/bluer/latest/bluer/struct.Adapter.html#method.set_pairable) works as well?

I apologize for the delay in my response. I just tested it, and it also works with set_pairable(false). Thanks!