bluez/bluer #140: Add a command line tool for device advertising

potto216, 2024-04-06:
What about adding a bluer command line tool for advertising from the command line? The purpose would be to allow users to easily implement wireless solutions such as hobbyists wanting to advertise sensor data from their Raspberry Pi. Current nondeprecated Bluez tools such as bluetoothctl cannot advertise from the command line. 

I have coded a very rough (not correctly formatted idea) at https://github.com/potto216/bluer-cmd/blob/main/le_advertise/src/main.rs

I think besides command line options the program should also be able to read advertising data from a JSON file so the user can write changes in the advertising data from the programing or scripting language of their choice.  

@surban what are your thoughts?

potto216, 2024-04-10:
Thanks, I'll work on those changes. I think an important use case is for the user to be able to change the advertising data if they need to update sensor values or available services. What about a command line option to exit the program after a specific time? They could then write a simple script loop that gives them the ability to change the advertisements with new command line arguments based on their changing conditions.

surban, 2024-04-25:
> What about a command line option to exit the program after a specific time? 

Yes, a timeout option seems reasonable, although not strictly necessary since people can just send a SIGTERM to the advertising process when scripting. Linux already [provides a command for that](https://man7.org/linux/man-pages/man1/timeout.1.html).

potto216, 2024-04-27:
At first, I liked the idea to either exit nicely based on SIGTERM or exit if the user presses ENTER. However, I ran into a problem if I received a SIGTERM and tried to cancel the STDIN, the program would still hang waiting for the ENTER (code in https://github.com/potto216/bluer-cmd/blob/main/le_advertise/src/main.rs). After some research I found out that a background task that reads from stdin in Tokio is blocking and impossible to kill nicely (ref: https://github.com/tokio-rs/tokio/discussions/5684 and https://docs.rs/tokio/latest/tokio/io/fn.stdin.html)

One solution is to forget about sending SIGTERM and have exit based on ENTER with stdin or use the below command to avoid `timeout` with `--kill-after`

`(sleep 5; echo) | cargo run -- -u 123e4567-e89b-12d3-a456-426614174000 -l bluer --discoverable --duration 3 -v`

To exit based on the conditions of another process we can use a named pipe. So for example,
```
# setup a named pipe
 mkfifo /tmp/my_fifo
# Have the advertiser listen on the pipe
cat /tmp/my_fifo | cargo run -- -u 123e4567-e89b-12d3-a456-426614174000 -l bluer --discoverable --duration 3 -v

# In another process stop advertising when the conditions are met
 echo > /tmp/my_fifo
```
@surban right now I'm leaning towards rolling back all the SIGTERM handling code and just document the above ways to interact with the le_advertise code. What do you think?

surban, 2024-04-29:
I would prefer the other way. Just tell the user "Press CTRL+C to stop advertising" and listen for SIGINT and SIGTERM. This will make it easy both for scripts and humans, chat bots, AIs, etc...

potto216, 2024-05-01:
I see your point and I'll modify the code to remove the stdio input and just use SIGINT and SIGTERM. One concern with just using SIGINT and SIGTERM is that it may not be obvious for people unfamiliar with Linux how to find the process id and send the signals. I think there should be documentation with common command line use cases. Possibly in a README markdown in the Bluer repo or as extended help with the tool.

surban, 2024-05-01:
I think we can assume a certain level of UNIX proficiency of our users, including process management.

potto216, 2024-05-01:
I think the Bluer command line tools are a fantastic way for people who have an idea they want to implement, for example that requires wireless and a Raspberry Pi, to create a solution without having to know much about Linux, Rust, or even Bluetooth. I can create some example gists and talk about them in the discussion's show and tell and see what response they get.

For the advertisement tool I was going to fork bluer and add the advertisement tool to the bluer-tools/src directory and call it perhaps--bluadv?

surban, 2024-05-02:
> I think the Bluer command line tools are a fantastic way for people who have an idea they want to implement, for example that requires wireless and a Raspberry Pi, to create a solution without [...]

I agree. 

I just don't want to extend documentation to the point that it becomes a tutorial since this will involve maintenance burden in the future. Also Rust is [evolving to support scripting](https://github.com/rust-lang/cargo/issues/12207), so such things could be implemented directly using the BlueR API without the command line tools in the future.

I think the best place for detailed instructions and usage examples would be a blog or personal homepage. An example project reading out a temperature sensor with source code etc. is probably worth more than 1000 documentation words.

> For the advertisement tool I was going to fork bluer and add the advertisement tool to the bluer-tools/src directory and call it perhaps--bluadv?

Sounds good to me!

PR #143 (2024-06-13): Add a command line tool for device advertising #140
Add a command line tool for device advertising. Closes #140
commit 8cbc669b (2024-06-13): bluadv: add a command line tool for LE advertising (#143)

Implement command line tool to broadcast Bluetooth LE advertisements.

Fixes #140.

---------
closed 2024-06-13 by surban