bluez/bluer #158: ReverseServiceDiscovery problem

Frank-Buss, 2024-11-27:
I ran into a problem with the ble-peripheral-rust library, which uses bluer. If I use the default setting of ReverseServiceDiscovery in /etc/bluetooth/main.conf on Linux, which is "true", then an Android device connecting to my Linux desktop wants to pair with it with a pin. See this issue: https://github.com/rohitsangwan01/ble-peripheral-rust/issues/1

Is it possible without changing the conf file to disable the ReverseServiceDiscovery with bluer? I searched the github repo for "reverse", but didn't find anything.

surban, 2024-11-27:
No, this is handled by `bluetoothd` and currently it does not expose a D-Bus interface to change this setting.

To get this implemented you would have to: 
1. Submit a patch to `bluetoothd` that exposes this functionality via D-Bus. This might be challenging since it is a system-wide setting and you would need to handle the case when different clients want conflicting settings.
2. Submit a PR to BlueR to use the new setting. This would be trivial.

Frank-Buss, 2024-11-27:
ok, thanks, I guess they might have a reason to define it system wide, maybe better not to make it changeable then

closed 2024-11-27 by Frank-Buss