bluez/bluer #164: Adapter support for Coded PHY

ricoheller, 2025-02-06:
Hello,

Is it possible to enable and use **LE Coded PHY** with bluer adapter?
If not, do you have any plans do support it?

Today I use *hcitool* to enable it:

```bash
$ hcitool cmd 08 31 03 07 07
```

Thank you

surban, 2025-02-10:
You can set [`secondary_channel` on your `Advertisement`](https://docs.rs/bluer/latest/bluer/adv/struct.Advertisement.html#structfield.secondary_channel) accordingly. I have never tested it though, since my adapter does not support it.

ricoheller, 2025-02-10:
Hello surban,

Sorry, I didn't express myself clearly.

I would like to use bluer to capture advertising messages from other devices that use BLE CODED PHY.
CODED PHY is not enabled by default in the adapter so I need to send the hci command first to enable it.

surban, 2025-02-10:
I am not aware of any interface provided by `bluetoothd` for user space to configure the PHY that BlueR could use.

Generally, `bluetoothd` should auto-configure the HCIs and enable supported features. Talking to the HCI directly and bypassing `bluetoothd` is generally not a supported scenario, since 1. it requires root privileges and 2. multiple processes would infer with each other.

You could ask on the bluez Slack and/or mailing list if there is any support and about the recommended way to implement it.